juchong/ADIS16448-RoboRIO-Driver #3: could not find ADIS16448

floogulinc, 2016-04-04:
When trying to use the java library, it fails to initialize giving this error to the driver station `ERROR 1 could not find ADIS16448`. I tried making it delay a bit before initializing it but it still doesn't work. The green light is on the unit and it appears to be connected.

juchong, 2016-04-04:
Hi! I would check the connector on the RoboRIO and on the IMU itself to
make sure none of the pins are offset. When initializing, the MXP SPI port
should be selected.

Let me know if this helps!
-Juan

floogulinc, 2016-04-04:
@juchong it is installed correctly and screwed in. After using a tweak of the code to give the SPI a little more time to work (https://github.com/FRC5333/2016-Stronghold/blob/master/src%2Fmain%2Fjava%2Ffrc%2Fteam5333%2Flib%2Fdevice%2FADIS16448_IMU.java#L110-L162) it does not give the same error most of the time but the IMU still doesn't function. It may be a hardware issue though.

floogulinc, 2016-04-04:
Update: With that fix i gave, it seems to initialize ok. But now the gyro display on smartdashboard that used to show direction on the ground now seems to show angle perpendicular to the ground.

juchong, 2016-04-04:
I don't think this feature has been built into the Java version of the
driver, but the sensor was initially written with the RoboRIO in the
horizontal orientation (Z axis going through the sensor's PCB). The LabVIEW
version of the driver allows for selecting the axis used for calculations.

floogulinc, 2016-04-04:
Yeah, it seems the Java version is setup for the Y axis to be the one going through the PCB, but it's actually the Z axis. Also, the delay with initializing is still an issue. i have to manually go and "Reboot Robot Code" in the dashboard after starting or resetting the robot to get the imu to work and I have the fix in the above posts as well as a 6 second delay.

juchong, 2016-07-09:
Unfortunately the Java/C++ code has not been updated to incorporate all the LabVIEW updates. I'm currently working with others in the community to update the driver.

closed 2017-01-15 by juchong

alice15311, 2020-02-16:
I also have something to say.