juchong/ADIS16448-RoboRIO-Driver #7: Drift Calibration

brhea, 2016-08-02:
Is it really appropriate to store a drift calibration and use it every time. The ADIS16448 seems to drift less than other chips that we've used and I thought the purpose of calculating a new drift calibration every time the robots are rebooted is to get a current calibration. I believe that the factory calibrates each chip and stores that calibration on chip. Storing a drift calibration and using the same one seems the same as the factory calibration.

My impression was that these chips have a different calibration every time they are started up and that it changes with the temperature, amount of ferrous metal in the surrounding building, etc. 

I'm aware that if you don't perform the new drift calibration provided, the driver will behave as before and perform a calibration every time it starts. We would not want to lose this capability.

On another topic. Our team did not use the HUD last year because we had some issues in getting it to work reliably; we only used the axis variable. If you defined another item for the AHRS Algorithm, it could be set to "No HUD" and disable the filter and save some processor time. The algorithms seem to be pretty math intensive. If performing vision, etc. processor time is valuable.

juchong, 2016-08-03:
Hi brhea, 

The ADIS16448 IMU has much lower noise and does exhibit substantially less drift than other offerings. That being said, the factory calibration is meant to minimize orthogonality error (axis to axis misalignment) and changes in sensitivity (scale) across temperature. Unfortunately, no calibration can compensate for gyro drift in its entirety, but merely reduce as many sources of error as possible. 

The "calibration" program included with the driver is perhaps a misnomer. Instead of calibration, this program should really be an "offset recorder". Using the default settings the program samples enough data to most effectively reduce drift (based upon the bias stability plot in the datasheet) and calculates an offset which is then applied to every sample. Even though a portion of the drift is due to uncontrollable noise sources, the drift characteristics will remain reasonably repeatable. Due to the nature of FRC and its short match times, this method of drift reduction is "good enough".

The 448 does have the ability to perform magnetometer soft and hard iron calibration, but both of these are unique to the application, mounting location, etc. and must be performed by the user. The 448, as with most MEMS and with the exception of the magnetometer, is mostly impervious to electromagnetic fields. As with any other electronics, given a high enough EM field strange things will begin to occur. 

The user has the ability to revert to the original "power-on offset calculation" method after recording offset data. The only thing that he/she would need to do is delete the calibration file using the RoboRIO web interface. 

I re-wrote the data-sampling portion of the driver and greatly improved CPU usage with some of NI's guidance, so I encourage you to give it a try! I'll work on incorporating an additional "Disable" case for those that are only using gyro integration. Both of the AHRS calculation algorithms should be fairly efficient, but "the proof is in the pudding". I'll add some comments to hopefully outline the performance trade-offs. 

Hope I've answered your questions!

-Juan

brhea, 2016-08-03:
Thanks for the detailed explanation. We'll give this a try and see how well it performs (it will probably be  next season before our team gets around to this). I'm really a fan of this chip as the performance is very much better than those we've used previously. 

The matches are relatively short at 2 1/2 minutes each but when we're developing/testing we'll run a bot for a couple of hours. In the past we've modified the drivers for gyros to continuously calibrate while the  robot is disabled,,, this minimized drift when running for long periods of time

closed 2017-01-11 by juchong