juchong/ADIS16448-RoboRIO-Driver #12: Accelerometer does not appear to be working w/ Java 2017 Version

niwhsa9, 2017-01-13:
Upon testing the sensor today, the gyro is working just fine and seems to be outputting the correct data. However, the accelerometer is not. Looking at all 3 axis, none of the accelerometers seem to be correctly reacting to our robot moving forward. I would guess that it is the Z axis that should be working, but it isn't. Additionally, the values of the accelerometer and constantly jumping around, they are never at 0.0 even when the robot is still but rather anywhere from 0.00-0.001. They never seem to exceed that upper bound even when the robot is moving.

closed 2017-01-14 by niwhsa9

juchong, 2017-01-15:
getAccelY() should be in g (gravity). Normal values when stationary should reflect X, Y near 0 and Z near 1 (assuming the sensor is oriented Z-down).  Note that the values will never stop "jumping around" due to inherent noise in the sensor. All sensors have this limitation regardless of technology, unfortunately. The values you're displayed are also likely decimated heavily in order to reduce bandwidth. This may give the illusion that the robot isn't experiencing high g's when in reality it is. I would encourage you to set up a data logger within the RoboRIO and plot the data in Excel to reproduce the robot's motion.

niwhsa9, 2017-01-15:
yup, I closed it because I just did exactly this today at practice. When
plotted in excel, it became clear immediately that I was wrong. I believe
what happened was that the SmartDashboard couldn't update fast enough (or
display enough precision) for me to acurately judge the data. Sorry for not
having checked this first, I didn't mean to waste your time! Thank you so
much for the help, my team and I really apreciate it.

juchong, 2017-01-15:
Great! No problem! If you don't mind me asking, what team are you with?

niwhsa9, 2017-01-15:
I'm with 2496. We have technically been around for a few years but all the
original mentors and members are gone and a new mentor started it back up
again this year so we are basically a begginer team with a bunch of complex
parts we don't know how to use xD!