juchong/ADIS16448-RoboRIO-Driver #14: [Suggestion] Accelerometer integration

602p, 2017-01-25:
What's the possibility of extending this (awesome!) library to do integration from the accelerometers?

I know it might have some big error issues, but I thought I'd bring it up, as it's something our team would be interested in.

Within the scope of the project?

juchong, 2017-01-25:
That would be a great addition! Once you have some tested code implemented, feel free to submit a pull request and I'll integrate it into the code.

niwhsa9, 2017-01-26:
Omg I'd appreciate this so much. We almost gave up on the accelerometer
because none of us know how to do the Kalman filter for accurate
integration.

niwhsa9, 2017-12-10:
juchong, I have some code for acelerometer integration that works reasonably well. It makes use of a median filter. Would you be open to adding it if I made a pull? It is not as accurate as a Kalman Filter but since there is currently no feature for this I figure its better than nothing. Let me know, if so I'll polish it off and pull.

juchong, 2017-12-10:
Hi!
I don't have a problem with integrating the accelerometer code. Is this for LabVIEW? If so, either of us can do the final integration. 

I don't have a complementary filter because the AHRS output should provide the user with a similar output. It wouldn't hurt to add, though.

niwhsa9, 2017-12-10:
Should've specified MB, this would be for Java and I could probably port it to C++ also. I don't know LabVIEW.

juchong, 2017-12-10:
Oh, that makes it easy then! 👍 

Go ahead and submit a pull request. I plan on updating the Java and C++ drivers to implement a few fixes I've already integrated into LabVIEW.

juchong, 2018-01-21:
Hi! I just upgraded the driver to support auto SPI, so now may be a good time to add any code you may be working on!

Thanks!