juchong/ADIS16448-RoboRIO-Driver #20: Update Java for 2018 WPILib

HeroCC, 2018-01-20:
Hello!
I'm trying to use this IMU code for the 2018 RoboRIO / WPILib, but it seems to depend on `wpilibJavaJNI` which no longer exists on the 2018 robot and causes the robot code to crash on startup. Is there any way to update this for the 2018 release of wpilib? 
Thanks!

Here is the full trace:
```
Exception in thread "main" java.lang.UnsatisfiedLinkError: no wpilibJavaJNI in java.library.path
        at java.lang.ClassLoader.loadLibrary(ClassLoader.java:1867)
        at java.lang.Runtime.loadLibrary0(Runtime.java:870)
        at java.lang.System.loadLibrary(System.java:1122)
        at edu.wpi.first.wpilibj.hal.JNIWrapper.<clinit>(JNIWrapper.java:22)
        at edu.wpi.first.wpilibj.RobotBase.initializeHardwareConfiguration(RobotBase.java:171)
        at edu.wpi.first.wpilibj.RobotBase.main(RobotBase.java:194)
```

juchong, 2018-01-21:
Hi! I just updated the Java driver to fix any compilation issues and upgrade the software to support "Auto SPI". Be sure to pull again!

HeroCC, 2018-01-21:
Yep, that works; thanks!

closed 2018-01-21 by HeroCC