juchong/ADIS16448-RoboRIO-Driver #21: C++ Errors

facepalmgamer, 2018-01-20:
My team and i have just set up out gyro and were messing with the smartdashboard, and as we deploy the code and enable our robot we get this error in the consol
[image]
so we checked the .h and it was fine but the ADIS16448_IMU.cpp in the src folder had 4 errors 2 being related to the the notify_all at line 239 and 328
[image]
the other 2 being related to line 368 where it says "wait" and line 961 where it says "ADIS16448_IMU::InitSendable"
[image]
[image]
does anyone else get these errors, if so how do we fix them because as of right now our gyro won't give us an angle.

facepalmgamer, 2018-01-20:
Sorry, I do not know why the picture are upside down.

juchong, 2018-01-21:
Hi there! It looks like you may not have re-indexed the database after opening the project. In Eclipse, right-click on the project and select Index > Rebuild". If you have the latest 2018 libraries installed, the errors should go away.

facepalmgamer, 2018-01-23:
as i'm deploying the code to the robot as well as when i'm activating it in the console i am receiving this error:
ERROR  -61072  Unknown error status  InitAuto [SPI.cpp:293]
	at frc::SPI::InitAuto(int)
	at ADIS16448_IMU::ADIS16448_IMU(ADIS16448_IMU::Axis, ADIS16448_IMU::AHRSAlgorithm)
	at Robot::RobotInit()
	at frc::TimedRobot::StartCompetition()
	at main
	at __libc_start_main

PeterJohnson, 2018-01-23:
InitAuto() is being called with 0 as the parameter.  This isn't valid; it looks like the Java code calls InitAuto() with 8200 as the parameter, and the C++ code should be changed to match.

juchong, 2018-01-23:
I'll check this as soon as I get home. I didn't realize the C++ code was throwing an error.

juchong, 2018-01-23:
Even after changing the value to 8200, the error is still triggered. Newest commit should reflect the change