juchong/ADIS16448-RoboRIO-Driver #24: Labview error

brhea, 2018-01-24:
I receive following from labview from Open vi.... I have latest copy as of Jan 15th. Kind of an odd error I would not expect. I don't think there is anything wrong with our code. Reads of the gyro seems to work. Error only occurs 1 time when code starts.

ERROR  1198  Invoke Node in ADIS16448_IMU.lvlib:ADI_IMU_ADIS16448_Open.vi:2260001->Drive Controller.vi->SubSystems.vi->Robot Main.viLabVIEW:  (Hex 0x4AE) The VI is not in a state compatible with this operation. Change the execution mode of the referenced VI to reentrant for this operation.

juchong, 2018-01-24:
I think I may know what's going on. Did you add the ADIS16448 LabVIEW project to a robot project, or did you only add the "ADIS16448 IMU" folder? If you added the project, did you attempt to open the example from within a robot project? If so, the issue could be that the example is now trying to reset the front panel, but code had already executed. LabVIEW is complaining about an invoke node not being able to be set, and the only place invoke nodes are used is in the example front panel. Hope this helps!

brhea, 2018-01-24:
Our team added the ADIS16448 IMU folder to the .../Thirdparty/Sensors folder so that the vi's show up in the palate. We then ported last years code to this year by replacing each of the vi's in last years code with this years vi's from the palatte. We deleted all of last years vi's so there is no possible conflict. 

I opened your "Open.vi" this morning and I think I may see the problem.... I also had a look at the example and it seems that the example does not use the "Loop.vi". My guess is that "Open.vi" is trying to run the "Loop.vi" and we're also trying to run the "Loop.vi" after "Open.vi" completes.... so "Loop.vi" is being called twice and it's not re-entrant (and shouldn't be). 

I think the new code below is what is causing the problem. This was not in Open last year.  I don't understand why this has been added. If the name of the WPI_DriverStationStart Communication.vi is ever changed it will break your code. I'm not certain why this was added. 

I'll be with our students later today and will remove the "Loop.vi" and see if the error goes away. I'll make another post if this is successful. 

[image]

juchong, 2018-01-24:
That portion of the code was added to help clean up overall robot code. The first while loop looks to see if driver station communication has been started, and if so, starts the IMU Loop.vi in the background. This makes it so that you don't have to call Loop.vi in your robot code. 

It sounds like the fix for your robot code is to not call Loop.vi anywhere else in your code.

brhea, 2018-01-25:
Thanks!.... I removed the "Loop.vi" and the error went away. 

Most sensors don't require that the driver's station be running before they operate. Some teams might find this confusing and waste a lot of time debugging before they figure out that they have to run the driver's station before their sensor will work. In competition, this has to run of course but not for initial testing of sensors. 

It might be appropriate to require that the "Device Found" flag be set before starting the "Loop.vi". There is no need to run it if the chip is not found and initialized. 

In previous years our team has had an occasional problem during autonomous mode at competitions where the bot boots but for some reason the ADIS chip was not found and the gyro was not functioning. This could be a timing issue where the roborio boots faster than the chip or a voltage dip causes the chip to reset during startup, etc. When this happened the bot tried to turn with the gyro to a position and it just spined for the duration of auto because the gyro didn't start. 

We added a while loop to your open.vi that sleeps for 1 second and retries to initialize the chip over and over again until "device found" is set. This got rid of the problem. Note that we had to close the interupt/SPI and DIO in before the retry. We had 0 failures of the gyro during competition last year after adding this code.

[image]

We'd like to be able to do something like the code below with your Open.vi and not have to modify the internals of the loop.... This would work if the VI completed without starting the "Loop.vi" and it also closed out all of the resources it uses if it failed to start. 

[image]

Note we also modified the typedef for the filter type and added "None" and then modified the loop adding a case structure so that we could disable the filter calculation completely if "None" was selected. This was to save processor time. We did vision processing on the roborio and tried to save every clock cycle possible. 

We'll probably make the modifications again this year if we have time to implement and test. 

Thanks for supporting this driver and for the quick response to issues! Keeping these up to date and supported is no small task!

juchong, 2018-01-25:
Hi!
I would **highly** suggest not adding a while loop around Open.vi. This year, the entire SPI communication methodology has been upgraded to use Automated SPI Transactions. Open.vi puts the RoboRIO SPI port in a state that will cause it to not respond to normal SPI transactions such as reading/writing registers from the part. Before you make any modifications, I would encourage you to try the driver as-is and submit any feedback if possible. 

As for requiring the driver station, unfortunately the requirement that the driver station be open comes from the way in which toggling DIOs was implemented. Because I'm required to restart the IMU every time the robot boots, I must toggle a DIO. **Toggling DIOs is restricted in the FPGA unless the driver station is open.** I've talked to NI about a workaround, but all of them would compromise safety. 

Again, I would highly suggest you try out the sensor as-is. Thinking back on your original issue, it sounds like your team was using a very old version of the driver. Maybe this version will solve your issue? If you have any questions, you can always email me at juan AT highcurrent.io

I'm going to close this issue since the original problem was resolved.

closed 2018-01-25 by juchong