juchong/ADIS16448-RoboRIO-Driver #25: Yaw delta moving 45 for every 90 degree turn

jasondaming, 2018-02-05:
We finally got the chance to use our ADIS16448 last night and we were seeing a constantish (always near it but it did "bounce around") 160 degrees on the yaw.  The yaw delta did however respond to the robot turning except it only moved 45 degrees for every 90 that the robot turned.

After seeing this we then went back and tried a calibration which worked fine and then tried the above again with the same results.

Any tips on why this might be doing this?  We are using LV if that matters with the same inputs to the open as the example.

brhea, 2018-02-19:
Our team had the same issue only we just use the Z axis. We multiplied by 2 until the driver is fixed.

jasondaming, 2018-02-23:
Are there any thoughts on this?  Seems like it is not just us and a firmware update to v17 didn't help.  I can't imagine that we are running the example wrong this sounds like a very basic driver issue

juchong, 2018-02-23:
The newest revision of the software should fix the issue. It looks like I used the incorrect delta-time value in the angle calculation. A new installer should be available in a bit.

closed 2018-02-23 by juchong