juchong/ADIS16448-RoboRIO-Driver #26: gyro found by roborio but no data is being gathered by gyro

TheBetterProgrammer, 2018-02-23:
we are having an issue where the ADIS16448 is apparently being found on the roborio (at least there is no gyro not found error), but no data is actually being sent or collected by the gyro.  any command to get angle x, y, z, or acceleration results in 0.0.  is seems like after several hours of re-seating the ADIS16448, swapping to a different ADIS16448, rebooting the robot, and resetting the code fixes the issue, but there is a chance that every time the robot is re powered the gyro once again does not collect and send any data.  do you have any clue as to what might be happening here?

juchong, 2018-02-23:
Hi, I'm not sure where to begin.. what programming language are you using? Is there anything that you've identified that will "always" make the sensor work?

Oldgoer, 2018-02-24:
We are using c++.  What i have been finding is that it appears that we have to swap to different ADIS16448s in order to get it to eventually work (without a swap it appears to never fix).  I was thinking that we perhaps had a failed device, but i have marked and gotten all three of ours running after swapping back and fourth.  On Saturday our team will be meeting again and i am planning on adding some additional console output code throughout the ADIS16448 cpp file to see what sections of the library we are hitting.