juchong/ADIS16448-RoboRIO-Driver #27: VSCode 2019 install

cpapplefamily, 2018-11-17:
I recall using this device years ago and manually having to put effort in to get this working. I'm Beta Testing the 2019 WPILib suites and associated FRC software.  Specifically Java. 

- Step One I was hung up on. There is no repositories block in my "build.gradle". I added line for line from your sample:
`
repositories {
  mavenCentral()
  maven { url 'https://jitpack.io' }
}
`

- Step Two was fine only my dependencies looks like this:
`dependencies {
    compile wpilib()
    compile javaVendorLibraries()
    nativeZip jniRoboRIOVendorLibraries()
    nativeDesktopZip jniDesktopVendorLibraries()
    testCompile 'junit:junit:4.12'
    compile 'com.github.juchong:ADIS16448-RoboRIO-Driver:master-SNAPSHOT'
}`

- Step Three. I don't know what you want here so I created the file tree under "src" com\analog\adis16448\frc and placed the "ADIS16448_IMU.java" in it.

- The import is Errored: [Java] The import com.analog can not be resolved

- The Make a new instance of the driver is Errored: [Java] ADIS16448_IMU Can not be resoved to type

MB3hel, 2018-11-18:
Hi @cpapplefamily 
It looks to me like you are trying to use gradle to add the library as a dependency, but have also copied it to source to the project directory (which is how the library is manually installed). You should only need to do one or the other and since you are using gradle adding it as dependency is recommended, however either way will work.

### To use the library as a gradle dependency:
First remove the sources you copied to `{PROJECT}/src`. This will make sure that there are no unnecessary files floating around in your project. 
It looks like there is an issue when using master-SNAPSHOT with jitpack. For now, using 2018.3 will work 

Add the following line in the dependencies section (instead of `compile 'com.github.juchong:ADIS16448-RoboRIO-Driver:master-SNAPSHOT'`)
```
implementation 'com.github.juchong:ADIS16448-RoboRIO-Driver:2018.3'
```
I used `implementation` here instead of `compile` because `compile` is deprecated and will be removed in gradle 3.0.

### If you would prefer to install the library manually:
Instead of copying the source file (`ADIS16448_IMU.java`) to `{PROJECT}/src/` copy it to `{PROJECT}/src/main`. This will add it as a part of the source set called "main". Gradle uses different source sets for different components of a project. For a FRC robot project, there is only one component: the robot code. Thus, the robot code is in the main source set. Placing the library in the `{PROJECT}/src/main` folder will make it a part the main source set.

cpapplefamily, 2018-11-18:
I tried your few suggestions manly removing the `{PROJECT}/src` and changed to `implementation 'com.github.juchong:ADIS16448-RoboRIO-Driver:2018.3'`

FRC and the WPILIB's are using VSCode this season with gradle. I believe we are at V 4.9
build.gradle has a line:
`wrapper {
     gradleVersion = '4.9'
}`

If the README.md is wrong we should get it fixed. 

Is it a requirement to be ONLINE only the first time the code is built?

MB3hel, 2018-11-18:
It will be required to have internet access the first time the code is built on any given system.

Gradle dependencies are pulled offline from different repositories and cached for use at later points in time. This is a feature of the Gradle build system. See the link below for more info.
https://docs.gradle.org/current/userguide/declaring_dependencies.html

The readme is meant to provide and example for how to use this library with the most recent stable setup (last year's stuff), so it has not been updated yet. Once I get to testing some of this for my team this year I will update it with a full example project for both C++ and Java using the 2019 build system.

As for the Gradle version: What I meant is that `compile` was deprecated and active maintenance was removed in Gradle 3.0.

cpapplefamily, 2018-11-18:
Thank you for your time and assistance. I'm  a user in this area and am learning more every day. I'll pass my newly acquired knowledge to others.  For the masses we will not read the complete master library to Gradle but knowing that when i turn the key key the motor gets fuel and runs its sometimes enough. 

The CTRE and NavX libraries have offline installers. That added to the manage offline vendors library WPILib options. Is this something that could be useful here? Or is the use case small enough to require an online connection and manually edits to the build.gradle file? 

For now my projects build. I'll plug in the hardware Monday night.

ThadHouse, 2018-11-18:
For future note, `implementation` does not properly work with the gradlerio setup. It still depends on `compile` and thats what any user needs to use.

MB3hel, 2018-11-19:
Ok, I was not aware of this. @cpapplefamily it looks like you should be using `compile` (see above comment). Sorry for the confusion.

MB3hel, 2019-01-15:
Closing this issue as there has been no activity recently and the new distribution method (as a vendor library) makes this irrelevant.

closed 2019-01-15 by MB3hel