juchong/ADIS16448-RoboRIO-Driver #29: Jitpack snapshots incompatible with GradleRio 2019.1.1

dannytix, 2019-01-10:
Jitpack has not produced a new snapshot since 2017.1.1, which is not compatible with the 'kickoff' release of GradleRio 2019.1.1 due to conflicts with OpenCV versions. Using Jitpack snapshots as described in the README will successfully compile, but will fail to run on the RoboRIO due to a missing opencv_java320 native dependency (the kickoff release moved to OpenCV 3.4.4.)

Jitpack snapshots are failing due to problems in the C++ build. There is a problem in the build.gradle file (addressed by [PR #28 ](https://github.com/juchong/ADIS16448-RoboRIO-Driver/pull/28) but the overall build structure needs further work before it will compile on Jitpack. C++ builds with GradleRIO now require the 'installRoboRioToolchain' task to be run before the build.

I don't think I will have time to tackle the C++ build issues myself this season, so I am submitting this as a warning to other FRC teams until someone else can pick it up. Team 568 only uses the Java library, so for now we've created our own Java-only fork. You can add this to your builds by replacing the dependency line in this project's README with:
compile 'com.github.FRC-568:ADIS16448-RoboRIO-Driver:java-SNAPSHOT'

MB3hel, 2019-01-11:
Thanks for letting me know. I will start working on updating the build system so to fix this now that the tool to automate running updateNIDrivers is complete.

MB3hel, 2019-01-15:
In order to be consistent with how other vendor libraries are distributed the ADIS16448 driver will be distributed as a vendor library using the newly defined system for this season (json and maven artifacts). As a result using jitpack is no longer supported and will not work. The documentation has been updated to explain how install the library using Visual Studio Code's Manage Vendor Libraries feature.

closed 2019-01-16 by juchong