juchong/ADIS16448-RoboRIO-Driver #33: Version mismatch

brhea, 2019-02-10:
Our team is using lab view and have downloaded latest version and installed. When deploying code we occasionally get an error indicating that the ADIS library on the bot does not match the project. Is this a problem. Gyro seems to work ok.

Drift is an issue at times. If we power down and back up there is little drift. We have open set to calibrate on open. However after testing for a while and disabling and re enabling multiple times drift may be very large 0.25 degree to 0.5 degree per second. 
Powering down seems to fix it for a while.

juchong, 2019-02-13:
@brhea Did you uninstall and re-install LabVIEW per the WPI instructions? If so, did you open a 2018 project using the 2019 libraries? I suspect that there's a version mismatch somewhere in your robot project that's causing these issues. As for drift, the calibration routine executes upon start-up, so every time that the robot code is restarted, you'll need to make sure the robot is stable.

brhea, 2019-02-14:
Yes we completely uninstalled labview before installing this years version. Then we started from a copy of the 2018 project. Is the solution to drag and drop each of the ADIS vi's from the 3rd party library and replace the ones in the project? 

If we completely uninstalled previous labview and then install the current ADIS libraries, shouldn't this use the latest library? 

We'll keep an eye on the drift. We have to run for a while before we start seeing this. Normally the bot is still when we boot it and do the calibrate.

juchong, 2019-02-15:
If you installed the library using the .exe installer, then the VIs should be present in the pallet. You may be running into issues with version mismatches if you also have a copy of the VIs in your robot project.

juchong, 2019-02-17:
Hi @brhea, I would suggest removing the library from your project and rely solely on the installer libraries. Please let us know if this solves your issue!

brhea, 2019-02-18:
I searched the entire drive on dev PC and there were some libraries in the 2018 project folder. I deleted these. I did not get any errors when loading the 2019 project so I think the vi's in the 2019 project were referencing the 3rd party library folder. I did not see the error over the weekend. I deployed for startup and also ran interactively.

brhea, 2019-02-19:
I think I found the real problem... Our students had failed to upgrade one of our PC's to the Season release vs. the preseason release of the drivers. Downloading from multiple pc's occasionally put an older verson of the driver on the roborio. This was somewhat random and was not consistent. 

I did note one issue with the driver in testing. We cannot use the "close" vi and then reopen the gyro with open. I was trying to set this up to minimize our down time during testing. The gyro never restarts if we close it and then try to reopen it. 

We have to redeploy the code or reboot the roborio. This is not an issue for competitions but after we test for a few minutes we notice that the gyro drift grows exponentially and gives us significant errors.
 
Occasionally on startup we also get larger drifts. This is random and probably due to the noise found during the calibration on startup. We are careful to not move the bot when starting up.

closed 2019-02-22 by brhea

juchong, 2019-02-22:
Hi @brhea, the restart is there for a reason. When code starts, the IMU is forced to reset to ensure it’s being configured in a known state. For reliability, I wouldn’t omit the reset.