juchong/ADIS16448-RoboRIO-Driver #36: Extreme gyro drift

316JakeMorris, 2019-02-17:
After pulling my hair out for a few days I have decided to open an issue. We have the ADIS 16448 installed on the MXP port on the rio through a More Board. We are programming in labview. The gyro will drift enormously in a small amount of time, sometimes as much as a degree a second. However, other times the gyro will barely drift at all and act fine. Any help would be greatly appreciated, thank you in advance.

juchong, 2019-02-17:
Hi! It sounds like the IMU is calibrating in lots of error on start-up. The sensor performs an offset calibration when robot code is first executed by default, so if your robot is moving during this time, you'll also calibrate in lots of error. Our IMU guide describes the behavior in more detail, but if you have any additional questions, please let me know! Here's a link to the guide: https://wiki.analog.com/first/

316JakeMorris, 2019-02-17:
Hello, I think I've actually solved the problem. I reinstalled the library (for the third time) and that seemed to do the trick. The robot was always stationary during start-up, yet after reinstalling the library the drift has cut down enormously. It's still not at levels specified in the data sheet, but small enough to not affect the robot in a 2 min and 30 sec match. I'll let you know if the issue pops up again, thanks so much!

closed 2019-02-17 by 316JakeMorris