juchong/ADIS16448-RoboRIO-Driver #37: Labview deployed vs interactive

brhea, 2019-02-22:
Just before we put our robot in the back we created a deploy for run on startup using labview and the driverstation messages indicated that the IMU16488 could not be found. We rebooted the robot several times and each time it could not be found.

When we use labview interactively, the IMU is found every time and works fine. 

We tried the same on another bot and the IMU was found whether running interactively or running from startup. 

Any experience with this. We're going to be a bit under the gun to troubleshoot this on inspection day.

juchong, 2019-02-22:
Hi, I’ve never seen this behavior before with regard to the IMU, but I have seen odd behavior with building and deploying code. This usually happens when the libraries on the PC building the image are corrupt, or there’s a mismatch between the Rio image and the libraries being loaded. Have you tried building your code on another PC?

brhea, 2019-03-04:
We updated our software using the latest FRC update at the Palmetto regional when we took the robot out of the bag. We had the same issue with the latest software. The IMU worked fine when using labview interactively but not when code was built and deployed for startup.

We swapped out the IMU for a spare one and it worked perfectly. It's possible that our IMU was one of the first ones that became available to FRC and is of an early revision and the firmware is not the same as the one that we swapped it for.

Is it possible to update the firmware on this chip to the latest level? I suspect that the chip is good but the communications firmware has a compatibility issue.

closed 2020-01-13 by juchong