juchong/ADIS16448-RoboRIO-Driver #41: Unhandled exception: java.lang.NoSuchMethodError: 'void edu.wpi.first.wpilibj.SPI.free()'

david-lemasurier, 2020-01-05:
Running the simulator from vscode results in the following error:
Simulator GUI Initialized!
HAL Extensions: Successfully loaded extension
********** Robot program starting **********
Error at com.analog.adis16448.frc.ADIS16448_IMU.<init>(ADIS16448_IMU.java:280): Unhandled exception instantiating robot com.analog.adis16448.frc.ADIS16448_IMU java.lang.NoSuchMethodError: 'void edu.wpi.first.wpilibj.SPI.free()'
        at com.analog.adis16448.frc.ADIS16448_IMU.<init>(ADIS16448_IMU.java:280)
        at com.analog.adis16448.frc.ADIS16448_IMU.<init>(ADIS16448_IMU.java:350)
        at frc.robot.Robot.<init>(Robot.java:30)
        at edu.wpi.first.wpilibj.RobotBase.runRobot(RobotBase.java:231)
        at edu.wpi.first.wpilibj.RobotBase.lambda$startRobot$0(RobotBase.java:329)
        at java.base/java.lang.Thread.run(Thread.java:834)

Warning at edu.wpi.first.wpilibj.RobotBase.runRobot(RobotBase.java:244): Robots should not quit, but yours did!
Error at edu.wpi.first.wpilibj.RobotBase.runRobot(RobotBase.java:245): Could not instantiate robot com.analog.adis16448.frc.ADIS16448_IMU!

PeterJohnson, 2020-01-05:
That will happen on the real robot as well, as that changed from 2019 to 2020.  #39 needs to be merged and a new release made to fix this.

juchong, 2020-01-12:
Hi,
The maven repository should be up-to-date with the latest features and fixes. Please let us know if you run into any additional issues!

closed 2020-01-12 by juchong