juchong/ADIS16448-RoboRIO-Driver #44: Gyro is not resetting

Vandit13, 2020-01-11:
Using the 2020 WPI code for FRC in java the reset method is not setting the gyro angle back to 0.

Here is my code

private ADIS16448_IMU imu = new ADIS16448_IMU();

  /**
   * This function is run when the robot is first started up and should be used for any
   * initialization code.
   */
  @Override
  public void robotInit() {
    // Instantiate our RobotContainer.  This will perform all our button bindings, and put our
    // autonomous chooser on the dashboard.
    m_robotContainer = new RobotContainer();
    imu.reset();
  }

  /**
   * This function is called every robot packet, no matter the mode. Use this for items like
   * diagnostics that you want ran during disabled, autonomous, teleoperated and test.
   *
   * <p>This runs after the mode specific periodic functions, but before
   * LiveWindow and SmartDashboard integrated updating.
   */
  @Override
  public void robotPeriodic() {
    // Runs the Scheduler.  This is responsible for polling buttons, adding newly-scheduled
    // commands, running already-scheduled commands, removing finished or interrupted commands,
    // and running subsystem periodic() methods.  This must be called from the robot's periodic
    // block in order for anything in the Command-based framework to work.
    CommandScheduler.getInstance().run();
    SmartDashboard.putNumber("imu", imu.getAngle());
  }

juchong, 2020-01-11:
The Java version of the 448 driver is still in disrepair. I'm currently working on getting it fixed by tonight. I rolled out updates to the C++ version last night, but ran out of time to port the code over to Java. I'll update this issue when I push the tested code. Sorry for the delay!

juchong, 2020-01-12:
Hi! The Java version of the driver should be good to go! Please let me know if you run into issues! You can add the library to your project by adding this URL as a vendor library: http://maven.highcurrent.io/vendordeps/ADIS16448.json

closed 2020-01-12 by juchong