juchong/ADIS16448-RoboRIO-Driver #47: Can you move the artifact to a secure repository?

swalberg, 2022-01-11:
Hi, the 2022 release of WPILib uses Gradle 7 which doesn't allow http based Maven repos. Is there a way you could use a repository that supports https, or enable https on maven.highcurrent.io? It has a certificate but it's self-signed.

Thanks,

Sean

ThadHouse, 2022-01-11:
This is now included in wpilib, so it no longer is a vendor deps. You can just import the adis class directly.

Do note that in the kickoff release it doesn't work, but that is all spi gyros. We hope to have a fix soon.

swalberg, 2022-01-11:
Oh nice, thanks! Is there a bug I can follow in wpilib?

ThadHouse, 2022-01-11:
It's in the known issues on the docs, but we don't have an explicit issue in the repo. It's actually a bug in the fpga image, so affects all languages.

juchong, 2022-01-11:
@ThadHouse It's worth pointing out that ADI does **not** own this software (so I hope the libraries were imported with the correct licensing and attribution!) I will **not** hide the repository, but will include a disclaimer and publicly archive it.

juchong, 2022-01-11:
@ThadHouse I pushed an update to both repositories that patches a significant flaw in the IMU start-up routine. The RoboRIO power supply (both V1 and V2) was designed to disable the 3.3V rail when VDD < 6V. This can lead to the IMU shutting down during a match. If this happens, the settings configured by the IMU driver at startup will be lost. The patched initialization routine will store the settings in the IMU flash, preventing heading loss during a match.